cifertech/ESP32-DIV #5: Compiler Error

zRCrackiiN, 2024-09-21:
C:\Users\User\Documents\Arduino\ESP32-DIV-main\Code\ESP32DIV\packetmonitor.cpp:11:1: error: 'arduinoFFT' does not name a type; did you mean 'ArduinoFFT'?
 arduinoFFT FFT = arduinoFFT();
 ^~~~~~~~~~
 ArduinoFFT
C:\Users\User\Documents\Arduino\ESP32-DIV-main\Code\ESP32DIV\packetmonitor.cpp: In function 'void do_sampling_FFT()':
C:\Users\User\Documents\Arduino\ESP32-DIV-main\Code\ESP32DIV\packetmonitor.cpp:142:3: error: 'FFT' was not declared in this scope
   FFT.Windowing(vReal, samples, FFT_WIN_TYP_HAMMING, FFT_FORWARD);
   ^~~
C:\Users\User\Documents\Arduino\ESP32-DIV-main\Code\ESP32DIV\HCScreen.cpp: In member function 'void HCScreen::showIcon(uint8_t, uint8_t, const HCIcon*)':
C:\Users\User\Documents\Arduino\ESP32-DIV-main\Code\ESP32DIV\HCScreen.cpp:226:22: error: 'class Adafruit_ST7735' has no member named 'Color565'; did you mean 'color565'?
         color = _tft.Color565(icon->pixel_data[idx++], icon->pixel_data[idx++], icon->pixel_data[idx++]);
                      ^~~~~~~~
                      color565
Multiple libraries were found for "SD.h"
  Used: C:\Users\User\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.9\libraries\SD
  Not used: C:\Users\User\AppData\Local\Arduino15\libraries\SD
  Not used: C:\Users\User\Documents\Arduino\libraries\SD
exit status 1

Compilation error: 'arduinoFFT' does not name a type; did you mean 'ArduinoFFT'?

zRCrackiiN, 2024-09-21:
Update

change arduinoFFT version to 1.3

Omd5144, 2024-10-14:
D:\arduino\ESP32-DIV-main\Code\ESP32DIV\packetmonitor.cpp:34:36: error: 'system_event_t' has not been declared
   34 | esp_err_t event_handler(void* ctx, system_event_t* event) {
      |                                    ^~~~~~~~~~~~~~
D:\arduino\ESP32-DIV-main\Code\ESP32DIV\packetmonitor.cpp: In function 'void packetmonitorSetup()':
D:\arduino\ESP32-DIV-main\Code\ESP32DIV\packetmonitor.cpp:228:3: error: 'tcpip_adapter_init' was not declared in this scope
  228 |   tcpip_adapter_init();
      |   ^~~~~~~~~~~~~~~~~~
In file included from C:\Users\User\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-33fbade6\esp32/include/esp_hw_support/include/esp_intr_alloc.h:11,
                 from C:\Users\User\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-33fbade6\esp32/include/esp_hw_support/include/esp_cpu.h:20,
                 from C:\Users\User\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-33fbade6\esp32/include/esp_hw_support/include/spinlock.h:11,
                 from C:\Users\User\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-33fbade6\esp32/include/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos/portmacro.h:74,
                 from C:\Users\User\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-33fbade6\esp32/include/freertos/FreeRTOS-Kernel/include/freertos/portable.h:59,
                 from C:\Users\User\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-33fbade6\esp32/include/freertos/FreeRTOS-Kernel/include/freertos/FreeRTOS.h:71,
                 from C:\Users\User\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.5\cores\esp32/Arduino.h:33,
                 from D:\arduino\ESP32-DIV-main\Code\ESP32DIV\packetmonitor.cpp:6:
D:\arduino\ESP32-DIV-main\Code\ESP32DIV\packetmonitor.cpp:230:19: error: 'esp_event_loop_init' was not declared in this scope; did you mean 'esp_event_loop_run'?
  230 |   ESP_ERROR_CHECK(esp_event_loop_init(event_handler, NULL));
      |                   ^~~~~~~~~~~~~~~~~~~
C:\Users\User\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-33fbade6\esp32/include/esp_common/include/esp_err.h:118:30: note: in definition of macro 'ESP_ERROR_CHECK'
  118 |         esp_err_t err_rc_ = (x);                                        \
      |                              ^
D:\arduino\ESP32-DIV-main\Code\ESP32DIV\detector.cpp:42:37: error: 'system_event_t' has not been declared
   42 | esp_err_t event_handler1(void *ctx, system_event_t *event) {
      |                                     ^~~~~~~~~~~~~~
D:\arduino\ESP32-DIV-main\Code\ESP32DIV\detector.cpp: In function 'void detectorSetup()':
D:\arduino\ESP32-DIV-main\Code\ESP32DIV\detector.cpp:144:3: error: 'tcpip_adapter_init' was not declared in this scope
  144 |   tcpip_adapter_init();
      |   ^~~~~~~~~~~~~~~~~~
In file included from C:\Users\User\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-33fbade6\esp32/include/esp_hw_support/include/esp_intr_alloc.h:11,
                 from C:\Users\User\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-33fbade6\esp32/include/esp_hw_support/include/esp_cpu.h:20,
                 from C:\Users\User\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-33fbade6\esp32/include/esp_hw_support/include/spinlock.h:11,
                 from C:\Users\User\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-33fbade6\esp32/include/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos/portmacro.h:74,
                 from C:\Users\User\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-33fbade6\esp32/include/freertos/FreeRTOS-Kernel/include/freertos/portable.h:59,
                 from C:\Users\User\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-33fbade6\esp32/include/freertos/FreeRTOS-Kernel/include/freertos/FreeRTOS.h:71,
                 from C:\Users\User\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.5\cores\esp32/Arduino.h:33,
                 from D:\arduino\ESP32-DIV-main\Code\ESP32DIV\detector.cpp:6:
D:\arduino\ESP32-DIV-main\Code\ESP32DIV\detector.cpp:146:19: error: 'esp_event_loop_init' was not declared in this scope; did you mean 'esp_event_loop_run'?
  146 |   ESP_ERROR_CHECK(esp_event_loop_init(event_handler1, NULL));
      |                   ^~~~~~~~~~~~~~~~~~~
C:\Users\User\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-33fbade6\esp32/include/esp_common/include/esp_err.h:118:30: note: in definition of macro 'ESP_ERROR_CHECK'
  118 |         esp_err_t err_rc_ = (x);                                        \
      |                              ^
D:\arduino\ESP32-DIV-main\Code\ESP32DIV\HCScreen.cpp: In member function 'void HCScreen::showIcon(uint8_t, uint8_t, const HCIcon*)':
D:\arduino\ESP32-DIV-main\Code\ESP32DIV\HCScreen.cpp:226:22: error: 'class Adafruit_ST7735' has no member named 'Color565'; did you mean 'color565'?
  226 |         color = _tft.Color565(icon->pixel_data[idx++], icon->pixel_data[idx++], icon->pixel_data[idx++]);
      |                      ^~~~~~~~
      |                      color565
Multiple libraries were found for "SD.h"
  Used: C:\Users\User\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.5\libraries\SD
  Not used: C:\Users\User\AppData\Local\Arduino15\libraries\SD
  Not used: C:\Users\User\Documents\Arduino\libraries\SD
exit status 1

Compilation error: 'system_event_t' has not been declared

closed 2025-05-13 by cifertech