cifertech/ESP32-DIV #7: More detailed BOM

lunneswang, 2025-04-28:
Hello, can you provide a more detailed BOM again? There are some components that cannot be matched without specific specifications.

dustinwloring1988, 2025-05-01:
@lunneswang which ones are you having problems with I will help you out

silentfox2023, 2025-05-02:
The integration of the PCB board has been completed, and all hardware functions are normal. In fact, the BOM list is already very detailed. If you are not clear about the model of any component, you can leave a message here.

lunneswang, 2025-05-03:
@dustinwloring1988 
Is the model of LF33 LF33CDT-TR
Is the model of ESP32 ESP32-WROOM-32U-N4
Is the model of 100uF50V RVT1H101M0607
Due to the large number of factories and types of components in our area, some of the components read by BOM have very different specifications

WXLN, 2025-05-14:
> [@dustinwloring1988](https://github.com/dustinwloring1988) LF33 的型号是 LF33CDT-TR 吗？ ESP32 的型号是 ESP32-WROOM-32U-N4 吗？ 100uF50V 的型号是 RVT1H101M0607 吗？ 由于我们地区元器件工厂较多，种类较多，所以有些元器件 BOM 读取到的规格差异很大

ESP32 4M/16M Fine There is updated help information in the wiki。 100uf You can choose the 0805 specification. Voltage is not important in this circuit, 50V is sufficient

lunneswang, 2025-05-14:
> > [@dustinwloring1988](https://github.com/dustinwloring1988) LF33 的型号是 LF33CDT-TR 吗？ ESP32 的型号是 ESP32-WROOM-32U-N4 吗？ 100uF50V 的型号是 RVT1H101M0607 吗？ 由于我们地区元器件工厂较多，种类较多，所以有些元器件 BOM 读取到的规格差异很大
> 
> ESP32 4M/16M Fine There is updated help information in the wiki。 100uf You can choose the 0805 specification. Voltage is not important in this circuit, 50V is sufficientESP32 4M/16M 细节，维基上已更新帮助信息。100uf 可以选择 0805 规格。在这个电路中，电压不重要，50V 足够

I saw in the videos and pictures that many components were not attached, including 100uF50V

WXLN, 2025-05-14:
> > > [@dustinwloring1988](https://github.com/dustinwloring1988) LF33 的型号是 LF33CDT-TR 吗？ ESP32 的型号是 ESP32-WROOM-32U-N4 吗？ 100uF50V 的型号是 RVT1H101M0607 吗？
> > 
> > 
> > ESP32 4M/16M Fine wiki 中有更新的帮助信息。 100uf 可以选择 0805 规格。电压在这个电路中并不重要，50V就足够了ESP32 4M/16M细节，维基上已更新帮助信息。100uf可以选择0805规格。在这个电路中，电压不重要，50V足够
> 
> 我在视频和图片中看到很多组件没有连接，包括 100uF50V

All of them belong to filtering capacitors and can be left unwelded.

lunneswang, 2025-05-14:
> > > > [@dustinwloring1988](https://github.com/dustinwloring1988) LF33 的型号是 LF33CDT-TR 吗？ ESP32 的型号是 ESP32-WROOM-32U-N4 吗？ 100uF50V 的型号是 RVT1H101M0607 吗？
> > > 
> > > 
> > > ESP32 4M/16M Fine wiki 中有更新的帮助信息。 100uf 可以选择 0805 规格。电压在这个电路中并不重要，50V就足够了ESP32 4M/16M细节，维基上已更新帮助信息。100uf可以选择0805规格。在这个电路中，电压不重要，50V足够
> > 
> > 
> > 我在视频和图片中看到很多组件没有连接，包括 100uF50V
> 
> All of them belong to filtering capacitors and can be left unwelded.它们都属于滤波电容器，可以不焊接。

Is it enough to weld according to the pictures and videos? I can burn it normally, but will it affect the functionality

WXLN, 2025-05-14:
> > > > > [@dustinwloring1988](https://github.com/dustinwloring1988) LF33 的型号是 LF33CDT-TR 吗？ ESP32 的型号是 ESP32-WROOM-32U-N4 吗？ 100uF50V 的型号是 RVT1H101M0607 吗？
> > > > 
> > > > 
> > > > ESP32 4M/16M Fine wiki中有更新的帮助信息。100uf可以选择0805规格。电压在这个电路中并不重要，50V就足够了ESP32 4M/16M细节，维基上已更新帮助信息。100uf可以选择0805规格。在这个电路中，电压不重要，50V足够
> > > 
> > > 
> > > 我在视频和图片中看到很多组件没有连接，包括100uF50V
> > 
> > 
> > 都属于滤波电容，可以不焊接。它们都属于冷却剂，可以不焊接。
> 
> 按照图片和视频焊接就可以了吗？r25、r26、q3、c9、c10可以正常烧录，但是会不会影响功能？

Capacitors should be soldered whenever possible, and resistors should not be lacking.

WXLN, 2025-05-16:
> 您好，有人能分享一下 PCB 的制造细节吗？我正在尝试向 NextPCB 下单，我可以上传 GERBER 文件，但不确定 PCB 的制造选项是什么。NextPCB 会从 .zip 文件中获取详细信息，可以吗？提前谢谢了。

I can only give you a reference because I use jlc manufacturing

[image]
The size needs to be calculated according to the platform you ordered, and we are not using the same board.

[image]

silentfox2023, 2025-05-16:
Here is the BOM LIST I used, with details updated. Please note that this is the original version by the author, and it is a PCB with unmodified circuits.

[ESP32DIV-SHIELD-BOM.docx](https://github.com/user-attachments/files/20256166/ESP32DIV-SHIELD-BOM.docx)
[ESP32DIV-BOM.docx](https://github.com/user-attachments/files/20256165/ESP32DIV-BOM.docx)

Regarding the selection of ESP32, anything is acceptable, but the packages and antennas are different. Note that do not use the type with an onboard antenna. In addition, the chip should be ESP32-D0WD-V3.

silentfox2023, 2025-06-04:
There are design flaws in ESP32DIV, where pin conflicts cause module initialization failures and stability issues. It is recommended that everyone use it with caution, and we suggest redesigning the circuit board to ensure usability and stability.

Additionally, regarding production costs, since everyone uses different components, the costs vary. For my NRF24L01P, I used the AS01-SPIPX, and for the CC1101, I used the AS07-M1101S. Including the circuit board, ESP32U, and other components, the total cost is approximately $50.

BrainScript2018, 2025-06-22:
hi

Is there already a plan for a new design?
when will it be available?

best regards

urbannative, 2025-08-13:
Silent fox, can I ask what you mean by redesign the circuit board? and are users still seemingly having problems with the board? This is my first time that im about toattempt to do a build, please excuse me if I end up on here with a bunch of questions! Although, it was very surprising to hear that after all the hard work that has gone into the development of this project, that some users are having troubles, in the video (Swiss army knife) things looked to work great, and in the videos I just happened to see 2 days ago, documenting the prototyping build, I was truely amazed at all the work that has gone into getting this tool to the  stage it is at currently, and even though I still need to order all the parts and then assemble everything, I really wanted to express some love and appreciation to CiferTech , it has been super cool to watch and has inspired me get more interested in PCBs, but not just this project, im a fan of all your work, its awesome, THANK YOU for sharing more than just the outcome, thank you for sharing the process, and encouraging folks to build it themselves (eventhough I really wish I could just purchase one that is assembled, lol, because im afraid I'll get frustrated during the assembly and I'll end up with a pile of electronic that arnt working) 
Also, thank you to all the users who read the comments and reach out to help those of us that are not as skilled, or might not have as good of an understanding as you do. I really hope at the end of the day I end up with a functioning version of this tool! Cheers!
P.S. this is also my first post on github even though I've been utilizing (reading) it for years to study all things cyber!

vitiok41, 2025-08-19:
Good afternoon, what transistors and tantalum capacitors did you use?

vitiok41, 2025-08-21:
N-Header 10X2 какое растояние между пипнов