cifertech/ESP32-DIV #8: BOM and CPL files?

kazz2020, 2025-04-28:
hi is  there any chance to get bom and cpl files for jlcpcb?

kazz2020, 2025-04-30:
so there is no chance for this files or what?

dustinwloring1988, 2025-05-01:
@kazz2020 the BOM is in the PCB folder here they are converted to csv for you though. The other just remake the schematic along with this BOM and then generate the needed file. 

[ESP32DIV-BOM.csv](https://github.com/user-attachments/files/20001734/ESP32DIV-BOM.csv)
[ESP32DIV-SHIELD-BOM.csv](https://github.com/user-attachments/files/20001733/ESP32DIV-SHIELD-BOM.csv)

kazz2020, 2025-05-01:
> [@kazz2020](https://github.com/kazz2020) the BOM is in the PCB folder here they are converted to csv for you though. The other just remake the schematic along with this BOM and then generate the [...]

ive figured this out and place order already. i will see what will come :P

kazz2020, 2025-05-01:
https://a.aliexpress.com/_EwVImmQ is this screen correct?

dustinwloring1988, 2025-05-01:
looks right just make sure to get the touch one

WXLN, 2025-05-03:
If only there were the author's SCH source code, there would be no need to manually do it

[image]

[image]

kazz2020, 2025-05-03:
> If only there were the author's SCH source code, there would be no need to manually do it
> 
> [image]
> 
> [image]

I made it myself sooogood luck

Punkcake21, 2025-05-06:
> If only there were the author's SCH source code, there would be no need to manually do it

hey can you share the pcb file? thank you

WXLN, 2025-05-09:
> > 如果有作者的 SCH 源代码，就不需要手动做了
> 
> 嘿，你能分享一下PCB文件吗？谢谢

Ample food and clothing by working with our own hands.

[image]
[image]
If you can add silk screen text by yourself, that would be great

closed 2025-05-13 by cifertech