cifertech/ESP32-DIV #10: CYD support

v1ckdesigns, 2025-04-30:
Hi, I have managed to compile and push the code to my CYD2uSB device. It boots up , I see a splash screen, but as soon as it gets to main menu backlight led is off so I can barely see anything. Is there anything using GPIO21 ?

tegtheevi, 2025-05-01:
Wow bro how did you do that

v1ckdesigns, 2025-05-01:
It requires adjusting VSPI pins and tft_espi library user_setup.h. 
Btw, looks like pcf8574 was causing no backlight and gpio21 was always LOW because of it. Disabled it and it works. I still need to adjust the touch as it's mirrored

WXLN, 2025-05-01:
> 它需要调整 VSPI 引脚和 tft_espi 库的 user_setup.h 文件。 顺便说一句，pcf8574 似乎导致没有背光，因此 gpio21 一直处于低电平。禁用它后就可以正常工作了。我仍然需要调整触摸，因为它是镜像的。
There are too many errors. I hope you can share a fixed bin file or a fixed library

dustinwloring1988, 2025-05-01:
@v1ckdesigns I am interested in doing it this way as well. Did you make a custom shield for the 3x NRF24 and 1x CC1101?

v1ckdesigns, 2025-05-13:
> [@v1ckdesigns](https://github.com/v1ckdesigns) I am interested in doing it this way as well. Did you make a custom shield for the 3x NRF24 and 1x CC1101?

I did not do any custom shield, I just used a 2x24 header. Connected all 3 to same SPI pins but i used different CE,CSN GPIOS for each nrf.

It worked, all 3 NRFs are present and active. I managed to get my touch calibrated also, I can easily swap it with cc1101 which is great but now with 1.0.1 I get a blank screen 🤦

WXLN, 2025-05-13:
[image]
Could you have bought the wrong screen module.

v1ckdesigns, 2025-05-14:
CYD uses ST7789 . It's all good now. I updated tft_espi library and my config was removed. With few changes in the code CYD DIV 1.1.0 is up: 

[image]

closed 2025-05-18 by v1ckdesigns

ripad, 2025-07-29:
> CYD uses ST7789 . It's all good now. I updated tft_espi library and my config was removed. With few changes in the code CYD DIV 1.1.0 is up:
> 
> [image]

I used the CYD touchscreen and it stopped working after flashing the code, what should be changed?

Git-Jay13, 2026-02-14:
Would you mind posting your setup files or the BIN file so I can flash it? I've been trying for months but I'm still having trouble