cifertech/ESP32-DIV #11: Compilation error

123456yzs, 2025-05-01:
During the compilation process on my computer, I encountered an error as shown in the image. I don't know how to resolve it. Can you regenerate the code based on your original code so that it can be directly uploaded as firmware for users to easily upload and use?

[image]

[image]

silentfox2023, 2025-05-02:
using the library of ESP32 version 2.0.11. In addition, the manual soldering of the PCB has been completed, and the hardware is running normally.

WXLN, 2025-05-02:
You need to watch the entire video on Youtu. According to the video, the last configuration is ESP 32, and additional libraries are also required.
Of which arduinofft Installing directly on Arduino may result in errors. Here, I have used the following address https://github.com/DragonyFox/PROT_OS7_ADASUIT/tree/main/libraries/arduinoFFT  And then it was successfully compiled

WXLN, 2025-05-02:
[image]

123456yzs, 2025-05-07:
Could you please send me the compiled firmware via email? I would be extremely grateful if you could.

---- Replied Message ----
| From | ***@***.***> |
| Date | 05/02/2025 17:53 |
| To | ***@***.***> |
| Cc | ***@***.***>***@***.***> |
| Subject | Re: [cifertech/ESP32-DIV] Compilation error (Issue #11) |
WXLN left a comment (cifertech/ESP32-DIV#11)

MVIMG_20250502_175242.jpg (view on web)

cifertech, 2025-05-13:
Binary (.bin) files have been added to the repository. Follow the instructions provided in the ESP32-DIV Wiki.

All the information regarding the project can be found at [ESP32-DIV Wiki](https://github.com/cifertech/ESP32-DIV/wiki)

closed 2025-05-13 by cifertech