cifertech/ESP32-DIV #13: I have this error, Compilation error: 'class PCF8574' has no member named 'pinMode'. updated the PCF8574 library and nothing has changed, please help

AndriiTemnyj, 2025-05-01:
I have this error, Compilation error: 'class PCF8574' has no member named 'pinMode'.
updated the PCF8574 library and nothing has changed, please help

silentfox2023, 2025-05-02:
Make sure that your library is "PCF8574" by Renzo Mischianti. Other libraries are not valid.

AndriiTemnyj, 2025-05-02:
hi, replacing the PCF8574 library didn't help, everything is the same

AndriiTemnyj, 2025-05-03:
> Try downgrading the ESP32 library to version 2.0.11.

nothing works, can you make a bin file, thanks in advance

silentfox2023, 2025-05-04:
[ESP32U-ESP32DIV-PCB.zip](https://github.com/user-attachments/files/20025670/ESP32U-ESP32DIV-PCB.zip)

AndriiTemnyj, 2025-05-04:
Thank you, the bin file has been downloaded.

AndriiTemnyj, 2025-05-07:
hi, it's me again. the question is will it work on the esp 32 module
wroom32d instead of esp32u. and the second question: this bin file
edp32u-espDIV-pcb.zip is the download file instead of esp32-div for arduino
ide

AndriiTemnyj, 2025-05-07:
Esp32 wroom 32d -4mb

cifertech, 2025-05-13:
All the information regarding the project can be found at [ESP32-DIV Wiki](https://github.com/cifertech/ESP32-DIV/wiki)