cifertech/ESP32-DIV #14: The integration of the PCB board is completed, but several problems have been encountered.

silentfox2023, 2025-05-04:
Today, the following three problems occurred during use. Firstly, through the serial port monitor, it was found that the NRF24L01 module could not be initialized. Secondly, the programming could not be carried out normally, and the prompt was "The packet header is invalid during serial port communication." Thirdly, in the serial port monitor, the prompt was "The lock of ADC1 was not acquired first before releasing it."
Guys, please help me out! This project is really crucial to me. I would be extremely grateful if you could help me solve these problems! 

By the way, the second problem has been solved. That is, by using the parameters "esptool.py --before no_reset --after no_reset", it is possible to achieve reading and writing. However, this parameter needs to be run after entering the download mode. 

The relevant error information is as follows: 
1.After adding the echo code by myself, it shows "NRF24L01 Radio 1 2 3 initialization failed!"

2.The error message output by the serial port monitor is as follows:
-----------------------------------------------------------------------------------------------------
esptool -b 115200 flash_id
esptool.py v4.8.1
Found 2 serial ports
Serial port COM12
Connecting......................................
COM12 failed to connect: Failed to connect to Espressif device: Invalid head of packet (0x63): Possible serial noise or corruption.
-----------------------------------------------------------------------------------------------------

3.The error message output by the serial port monitor is as follows:
-----------------------------------------------------------------------------------------------------
rst:0x1 (POWERON_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13260
load:0x40080400,len:3028

csum err:0xcc!=0xae
ets_main.c 384 

ets Jul 29 2019 12:21:46
rst:0x10 (RTCWDT_RTC_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13260
load:0x40080400,len:3028
entry 0x400805e4

E (564) ADC: adc1_lock_release(419): adc1 lock release called before acquire
E (565) ADC: adc1_lock_release(419): adc1 lock release called before acquire
E (567) ADC: adc1_lock_release(419): adc1 lock release called before acquire
E (574) ADC: adc1_lock_release(419): adc1 lock release called before acquire
E (581) ADC: adc1_lock_release(419): adc1 lock release called before acquire
E (587) ADC: adc1_lock_release(419): adc1 lock release called before acquire
E (594) ADC: adc1_lock_release(419): adc1 lock release called before acquire
E (601) ADC: adc1_lock_release(419): adc1 lock release called before acquire
E (608) ADC: adc1_lock_release(419): adc1 lock release called before acquire
E (614) ADC: adc1_lock_release(419): adc1 lock release called before acquire

Button 0: Pressed
Button 1: Pressed
Button 2: Pressed
Button 3: Pressed
Button 4: Pressed
Button 5: Pressed
Button 6: Pressed
Button 7: Pressed
-----------------------------------------------------------------------------------------------------

WXLN, 2025-05-04:
你这应该是没有进入烧录模式，通电之前先按住boot键，通电以后按一下rst键，此时长按boot键，不要松手，1秒钟以后松开，然后再去烧录

You probably haven't entered the burning mode yet. Before powering on, press and hold the boot button. After powering on, press and hold the RST button. Don't let go of the boot button, release it after 1 second, and then proceed with burning

silentfox2023, 2025-05-04:
The second problem has been solved. That is, by using the parameters "esptool.py --before no_reset --after no_reset", it is possible to achieve reading and writing. However, this parameter needs to be run after entering the download mode.

WXLN, 2025-05-04:
> 第二个问题已经解决。也就是说，通过使用参数“esptool.py --before no_reset --after no_reset”可以实现读写。不过，该参数需要在进入下载模式后运行。

I haven't received my PCB yet, I will check if there are any similar issues once I receive it

[image]

silentfox2023, 2025-05-06:
> > 第二个问题已经解决。也就是说，通过使用参数“esptool.py --before no_reset --after no_reset”可以实现读写。不过，该参数需要在进入下载模式后运行。
> 
> I haven't received my PCB yet, I will check if there are any similar issues once I receive it
> 
> [image]

thanks!!

By the way. The components used are basically consistent with the BOM list. However, there are some uncertainties. For example, the actual appearance size of the 100uF 50V capacitor does not conform to the BOM. The CC1101 uses AS07-M1101S, and the NRF24L01P uses E01-2G4M27SX.

silentfox2023, 2025-05-07:
Regarding some issues, I have written the code myself and solved them by using the serial monitor for troubleshooting, recompiling, and flashing. This includes detecting the status and address of the PCF8574T module, as well as addressing the problem of "csum err:0xcc!=0xae, ets_main.c 384" during the startup process. Below is the content echoed by the serial monitor.  

--------------------------------------------------------------------------------------------------------------------------------
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1344
load:0x40078000,len:13936
load:0x40080400,len:3600
entry 0x400805f0
E (586) ADC: adc1_lock_release(419): adc1 lock release called before acquire
E (587) ADC: adc1_lock_release(419): adc1 lock release called before acquire
E (589) ADC: adc1_lock_release(419): adc1 lock release called before acquire
E (596) ADC: adc1_lock_release(419): adc1 lock release called before acquire
E (602) ADC: adc1_lock_release(419): adc1 lock release called before acquire
E (609) ADC: adc1_lock_release(419): adc1 lock release called before acquire
E (616) ADC: adc1_lock_release(419): adc1 lock release called before acquire

-------------------------------------------------------------------------------------------------------------------------------
ets_main.c 384 
ets Jul 29 2019 12:21:46

rst:0x10 (RTCWDT_RTC_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13260
load:0x40080400,len:3028
entry 0x400805e4

I2C Scanner
Scanning...
I2C device found at address 0x20
done

-------------------------------------------------------------------------------------------------------------------------------
Judging from the information output by the serial port, the I2C scanning program in the code has successfully detected the I2C device with the address of 0x20. This means that the PCF8574T is basically normal in terms of hardware connection and I2C communication. However, issues such as the previously occurred checksum error and ADC lock release error may still exist. It is suspected that there is a problem with the source code.

silentfox2023, 2025-05-07:
As you wish, and I have fixed some of the errors. Finally, I will attach several photos. I hope everyone will pay attention, and let's make ESP32DIV more perfect together.

[image]
[image]
[image]
[image]

123456yzs, 2025-05-07:
I sincerely hope to obtain the compiled .bin file from you, as I encountered errors while compiling the source code. Would you be willing to share the firmware with me? If so, please send it to me.

---- Replied Message ----
| From | ***@***.***> |
| Date | 05/07/2025 21:16 |
| To | ***@***.***> |
| Cc | ***@***.***>***@***.***> |
| Subject | Re: [cifertech/ESP32-DIV] The integration of the PCB board is completed, but several problems have been encountered. (Issue #14) |
silentfox2023 left a comment (cifertech/ESP32-DIV#14)

As you wish, and I have fixed some of the errors. Finally, I will attach several photos. I hope everyone will pay attention, and let's make ESP32DIV more perfect together.

***@***.***_8da51c00_5ecb374547fb86d98afae969790a42bb.mmweb_appid.wx_webfilehelper.jpg (view on web)
***@***.***_8da51c00_5ecb374547fb86d98afae969790a42bb.mmweb_appid.wx_webfilehelper.jpg (view on web)
***@***.***_8da51c00_5ecb374547fb86d98afae969790a42bb.mmweb_appid.wx_webfilehelper.jpg (view on web)
***@***.***_8da51c00_5ecb374547fb86d98afae969790a42bb.mmweb_appid.wx_webfilehelper.jpg (view on web)

123456yzs, 2025-05-08:
压缩包没看到呀，能不能麻烦你分享一下固件，万分感谢

---- 回复的原邮件 ----
| 发件人 | ***@***.***> |
| 日期 | 2025年05月08日 10:40 |
| 收件人 | ***@***.***> |
| 抄送至 | ***@***.***>***@***.***> |
| 主题 | Re: [cifertech/ESP32-DIV] The integration of the PCB board is completed, but several problems have been encountered. (Issue #14) |
WXLN left a comment (cifertech/ESP32-DIV#14)

我真诚地希望从您那里获得编译好的.bin文件，因为我在编译源代码时遇到了一些错误。您愿意分享固件给我吗？如果可以，请发给我。
…

往上翻，上面不是有个压缩包吗？

WXLN, 2025-05-08:
> 压缩包没看到呀，不能麻烦你分享一下，万分感谢
> […](#)

https://github.com/user-attachments/files/20085020/ESP32U-PCB-Modify_With_Problem.zip
也许你可以尝试捐献眼角膜

WXLN, 2025-05-08:
[image] 
I have also received the PCB I made myself and tried its functions, but it seems that it is not perfect. I'm afraid I need to develop it myself.. Only the left part can be used, the right part is blank.

123456yzs, 2025-05-08:
你是橘子大佬对吧？我看你发的图就是

---- Replied Message ----
| From | ***@***.***> |
| Date | 05/08/2025 21:12 |
| To | ***@***.***> |
| Cc | ***@***.***>***@***.***> |
| Subject | Re: [cifertech/ESP32-DIV] The integration of the PCB board is completed, but several problems have been encountered. (Issue #14) |
WXLN left a comment (cifertech/ESP32-DIV#14)

1000026167.jpg (view on web)
I have also received the PCB I made myself and tried its functions, but it seems that it is not perfect. I'm afraid I need to develop it myself.. Only the left part can be used, the right part is blank.

silentfox2023, 2025-05-09:
Great! So can the CC1101 module and the NRF24L01P module work properly?

WXLN, 2025-05-09:
> 太棒了！那么CC1101模块和NRF24L01P模块可以正常工作吗？

Sure, I can test everything. No code modifications were made

WXLN, 2025-05-09:
[image]
[image]
Currently, the application on the right is usable.

silentfox2023, 2025-05-10:
There may be differences in the libraries by Arduino. I sincerely request that you provide them. Thank you again.  ：）

WXLN, 2025-05-10:
> There may be differences in the libraries by Arduino. I sincerely request that you provide them. Thank you again. ：）

如说真有区别那也只有 arduinoFFT是2024年之前的旧版了。其他的没有区别

closed 2025-05-13 by cifertech