cifertech/ESP32-DIV #15: CPL FILE

Punkcake21, 2025-05-04:
Hey can someone share the CPL file?
Thanks

WXLN, 2025-05-04:
CPL ?

Electroteddy, 2025-05-11:
CPL = Pick&Place file for jlcpcb.com to place the components from the BOM in the correct position.

WXLN, 2025-05-12:
> CPL = jlcpcb.com 的 Pick&Place 文件，用于将 BOM 中的元件放置在正确的位置。 我也需要这个文件。

It seems that there is no such thing. I suggest making your own PCB schematic, which has already been verified. There's no big problem, you can do it yourself

[image]
[image]

joelpr1234, 2025-05-13:
> > CPL = jlcpcb.com 的 Pick&Place 文件，用于将 BOM 中的元件放置在正确的位置。 我也需要这个文件。
> 
> It seems that there is no such thing. I suggest making your own PCB schematic, which has already been verified. There's no big problem, you can do it yourself
> 
> [image] [image]

If they are asking, probably is because they don't know how to do that... I don't know how to do it, so instead of replying to everyone "you can do it", why you don't reply with the process to do it or just share the CPL file if you know how to do it?

Thanks.

WXLN, 2025-05-13:
> > > CPL = jlcpcb.com 的 Pick&Place 文件，用于将 BOM 中的元件放置在正确的位置。我也需要这个文件。
> > 
> > 
> > 好像没有这回事。建议你自己做个PCB原理图，已经验证过了。没什么大问题，自己做就可以了。
> > [image] ![图像](https://private-user-images.githubusercontent.com/127818672/442632488-cff86b3d-f189-40c3-8d6d-743871a78e55.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDcxNjAyMjcsIm5iZiI6MTc0NzE1OTkyNywicGF0aCI6Ii8xMjc4MTg2NzIvNDQyNjMyNDg4LWNmZjg2YjNkLWYxODktNDBjMy04ZDZkLTc0Mzg3MWE3OGU1NS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwNTEzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDUxM1QxODEyMDdaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0yNzQ4YTdjZTE2ZDBiZGQyYjE2MjEwMjMzNWQ2N2EyNDY0YWJmYjViNWVlNzg3ZDUxM2EyZjljZmI2NWQwMDIxJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.trWqbfPuG9dbHio1HLwEpOTVcgwRs8En5_Hb1Dtb2Ng)
> 
> 如果他们问了，可能是因为他们不知道怎么做...我不知道怎么做，所以你为什么不回复每个人“你可以做到”，而是回复操作过程或者如果你知道怎么做的话就分享 CPL 文件呢？
> 
> 谢谢。

How can we give it without such a thing? It's stupid to grow cotton?

WXLN, 2025-05-13:
Nothing can be achieved without effort, rather than simply seeking help. Even this thing doesn't exist in itself. Although I really want to help you, the author did not share this thing.

Punkcake21, 2025-05-13:
> Nothing can be achieved without effort, rather than simply seeking help. Even this thing doesn't exist in itself. Although I really want to help you, the author did not share this thing.

Maybe you can help by giving instruction for building the device🧐...
The CPL (Component Placement List) specifies the exact position, rotation and on which side of the board each component reference designator needs to be populated. Alternative names for the CPL File are 
-Pick and Place
-Centroid file
-XY file
-Position file
This typo file is required by JLCPCB for building PCBA (assembled).
So if (you said) none of this existed, can you tell us how you built yours since I have zero experience in PCB field.
Thank you

cifertech, 2025-05-13:
To anyone requesting CPL files: for obvious reasons (greedy companies abuse), I will not share them. I solder them by hand; you can do the same. Pick up your soldering iron and give it a try.

All the information regarding the project can be found at [ESP32-DIV Wiki](https://github.com/cifertech/ESP32-DIV/wiki)

closed 2025-05-13 by cifertech