cifertech/ESP32-DIV #18: Error ieee8211

WXLN, 2025-05-12:
According to the AI solution Add one at the front my_Although it can compile without any errors, it still cannot complete the compilation prompt text section exceeds available space in board It's really puzzling, the library files have already been put in

WXLN, 2025-05-12:
After commenting on the error location, the compilation can only pass normally！！I don't know what consequences it will cause

// Override Wi-Fi sanity check
//extern "C" int ieee80211_raw_frame_sanity_check(int32_t arg, int32_t arg2, int32_t arg3) {
//    return 0;
//}

cifertech, 2025-05-12:
Hi @WXLN 

Don't comment the codes, follow the instructions;
Here: https://github.com/cifertech/ESP32-DIV/wiki/Firmware-Upload#method-3-upload-sketch-via-arduino-ide

WXLN, 2025-05-13:
> 你好[@WXLN](https://github.com/WXLN)
> 
> 不要注释代码，按照说明操作； 这里：[https ://github.com/cifertech/ESP32-DIV/wiki/Firmware-Upload#method-3-upload-sketch-via-arduino-ide](https://github.com/cifertech/ESP32-DIV/wiki/Firmware-Upload#method-3-upload-sketch-via-arduino-ide)

Thank you, it is indeed a problem that has been solvedI didn't expect the wiki to have any explanation.