cifertech/ESP32-DIV #21: GPS Wardriving

Xyrusaur, 2025-05-17:
Out of curiosity, has anyone managed to implement GPS Wardiving with their div boards? I'm planning to make my own custom board, and any advice would be much appreciated. Thanks

WXLN, 2025-05-17:
Is this function available?

WXLN, 2025-05-17:
This board doesn't have this function, does it?

Xyrusaur, 2025-05-17:
No, it does not have wardriving built in. Just wondering if anyone has made their own modifications though

WXLN, 2025-05-17:
> 不，它没有内置驾驶功能。只是想知道是否有人自己做过修改。如有造成困扰，敬请谅解。

This project is too new and should not have been developed there yet.

silentfox2023, 2025-05-17:
In terms of hardware, it should be feasible. The SHIELD PCB has RX/TX pins reserved. In addition, there are two sets of loops for 3.3V and 5V, as well as I2C/SPI communication interfaces.

In fact, the Bruce project has integrated many powerful functions, including penetration capabilities in SubGHz, BLE, Wi-Fi and other aspects, but it also contains more. In particular, the Bruce Smoochiee version is much smaller than the ESP32DIV.

Xyrusaur, 2025-05-19:
Thanks for the feedback, I'll have a look into the bruce project and see if it's right for me.

closed 2025-05-19 by Xyrusaur