cifertech/ESP32-DIV #26: Feature_requeste

Musaiyaf, 2025-05-19:
Add ble wardriving and WiFi wardriving

Musaiyaf, 2025-05-25:
> it would really need GPS added for this...

Yes

zRCrackiiN, 2025-05-25:
started the functions, in my fork

Musaiyaf, 2025-05-25:
> started the functions, in my fork
 r u going to add wardriving

Musaiyaf, 2025-05-25:
> started the functions, in my fork

Are you going to add wardriving

zRCrackiiN, 2025-05-25:
That's what I've started, I've got the main functions in, it scans and logs
to sd card - ssid, rssi, channel and security, I still need to add hardware
integrations for a gps module

Musaiyaf, 2025-05-25:
> That's what I've started, I've got the main functions in, it scans and logs
> to sd card - ssid, rssi, channel and security, I still need to add hardware
> integrations for a gps module
> […](#)

What is the range of the module