cifertech/ESP32-DIV #28: Wiki Update needed

DanielR92, 2025-05-22:
Hello,

I found some small wrong values:
https://github.com/cifertech/ESP32-DIV/wiki/Firmware-Upload

Old: 
`Configure Flash Settings:

    Select ESP32 chip type and Developer Mode or Download Mode.
    Load the .bin file, set flash offset to 0x1000.
    Load the .partitions file, set flash offset to 0x8000.
    Set SPI Speed to 40 MHz, SPI Mode to DIO.
    Check the .bin and .partitions file boxes.`

New:
`Configure Flash Settings:

    Select ESP32 chip type and Developer Mode or Download Mode.
    Load the .bin file, set flash offset to 0x10000.
    Load the .partitions file, set flash offset to 0x8000.
    Set SPI Speed to 40 MHz, SPI Mode to DIO.
    Check the .bin and .partitions file boxes.`

I run into a issue with the "ESP32 Flash Downlaoder":
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 115200
Changed.
NO XMC flash  detected!
FLASH_CRYPT_CNT 0
ABS_DONE_0 False
Detected overlap at address: 0x8000 for file: C:\Users\xxxx\Desktop\flash_download_tool\dl_temp\bin_tmp\downloadPanel1\ESP32DIV.partitions.bin
[2025-05-22 20:54:41,167][ESP8266Loader_spi[1]][espDownloader.py][line:846][ERROR]: ESP32 bin file overlap error esp_write_flash.

DanielR92, 2025-05-23:
I know it is needed to have three files:

Offset     Path File
0x1000     bootloader.bin
0x8000     partition-table.bin
0x10000    firmware.bin

Or I am wrong here?

WXLN, 2025-05-24:
The pre compiled firmware is built for the LeXin ESP32-WROOM-32UE WiFi module. If you are not using the one shown in the picture or the s3 s2 series, you will need to modify the code and compile it yourself.

[image]

WXLN, 2025-05-24:
The pre compiled firmware is built for the LeXin ESP32-WROOM-32UE WiFi module. If you are not using the one shown in the picture or the s3 s2 series, you will need to modify the code and compile it yourself.

[image]
[image]

zRCrackiiN, 2025-05-25:
for a start, is the cp2101 recognized by your pc when you connect?

wfhstudio, 2025-05-27:
> I know it is needed to have three files:
> 
> Offset Path File 0x1000 bootloader.bin 0x8000 partition-table.bin 0x10000 firmware.bin
> 
> Or I am wrong here?

not need 3 files just use the one provided by cifer tech ... put the address as "0" (Zero)

zRCrackiiN, 2025-05-28:
Need to see some logs post them please

zRCrackiiN, 2025-05-29:
interesting that RTCWDT is RTC watchdog timer, it may be seeing a lock up of some sort on boot, 

could be brown out related or a bootstrap pin issue, send some really detailed photos of the built boards please, also try run it with the shield removed

Lyf-54, 2025-06-02:
Did you solve it, I also have a problem like you

DanielR92, 2025-06-05:
Hello everyone,

I now have the new ESP. But it's not working right now.
It fails during the flashing process; I keep getting the error message that the partition overlaps.

Like https://github.com/cifertech/ESP32-DIV/issues/33 here. It is the same.

DanielR92, 2025-06-05:
I use now esptool.exe

`.\esptool.exe -b 115200 write_flash --flash_size detect 0x8000 .\ESP32DIV.partitions.bin 0x1000 .\bootloader.bin 0x10000 .\ESP32DIV-v1.1.0.bin`

esptool.py v4.8.1
Found 2 serial ports
Serial port COM23
Connecting.....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting....
Detecting chip type... ESP32
Chip is ESP32-D0WD-V3 (revision v3.1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 70:b8:f6:79:1d:74
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Unexpected chip id in image. Expected 0 but value was 9. Is this image for a different chip model?

A fatal error occurred: .\bootloader.bin is not an ESP32 image. Use --force to flash anyway.