cifertech/ESP32-DIV #31: Pick N Place file?

t1nkersec, 2025-05-30:
Hi!

I'm trying to build this project with the PCBs! Is there currently any way to have the PnP file? 

It's not mentioned in the repository and is a requirement for printing the PCBs.

I'm not sure if Cifer is going to re-evaluate the current board as I think he mentioned there are some hardware bugs too at the moment (maybe he'll also finally add a BW16 as well! 😁)

Many thanks!

zRCrackiiN, 2025-06-03:
wont happen, you'll have to hand solder it!

gfranju, 2025-06-03:
I'm not good enough for SMD. I've never even tried it and to be honest I don't even have the necessary equipment. At my age...if I can solder without shaking it's a good day!