cifertech/ESP32-DIV #32: My CP2102 fried.

kazz2020, 2025-05-31:
Can some one check what could be wrong?

[image] 
when i connect to usb my cp2102 is super hot and LF33 to. Maybe something is wrong during pcb mount?

WXLN, 2025-05-31:
This kind of breakdown is usually caused by voltage overload，I suggest you use an ammeter to check the output voltage

kazz2020, 2025-05-31:
> This kind of breakdown is usually caused by voltage overload，I suggest you use an ammeter to check the output voltage

where exactly put metter probes?

LTD284, 2025-05-31:
i have the same problem lf33 is very hot someone please give me the problem

kazz2020, 2025-05-31:
Lf33 is hot but cp2102 is super hot and burned

kazz2020, 2025-06-01:
> Actually, I don't quite agree with the schematic provided, so I made it myself. I don't quite understand the switch part of the battery. So I adopted an automatic switching circuit.
> 
> [image]

[image]

Do you have different pcb? What should be changed in original pcb?

kazz2020, 2025-06-01:
> When the USB is connected, the CP2102 and LF33 become extremely hot and even burn out, and the pins of the LF33 emit a beeping sound in the conduction mode (indicating a possible short circuit between the pins). Remove the capacitors and resistors near the CP2102 and LF33, and test for short circuits (such as capacitor leakage or abnormal resistor values).
> It's all good now. The problem I had before was due to using cheap..... poor-quality components.fuck it.

So what exactly should I replace to make it work? Can you guide me to solve this issue? Mark on Pic and write values maybe? It would be cool😎

silentfox2023, 2025-06-02:
Since it needs measuring and making step-by-step judgments, I can't really help you with this. You should use a multimeter and follow the circuit diagram to check things like resistance, the voltage of chips, and whether the wires are connected properly. Hope it goes well for you!
By the way, it's possible that there is a problem with your soldering method. For example, during the heating process with solder paste and a hot air gun, applying too much solder paste may cause a short circuit between the pins.

kazz2020, 2025-06-02:
> Since it needs measuring and making step-by-step judgments, I can't really help you with this. You should use a multimeter and follow the circuit diagram to check things like resistance, the voltage [...]

But this board was soldered on jlcpcb. Bom took from GH so. I thought that there is some mistake on pcb from GH.

kazz2020, 2025-06-05:
Happy to soon. Now I tried to connect again and again my cp2102 is super hot. I dint get it

kazz2020, 2025-06-05:
> The GND at the bottom of CP2102 needs to be soldered, which requires the combination of solder paste and a hot air gun. Note that the amount of solder paste should not be too much. In addition, it is recommended that you take the time to read the circuit diagram and use a multimeter to measure the readings of each pin. Only in this way can you identify and solve the problem. Good luck.

So it was like this. I've order on jlcpcb and got two pcb mounted. I had to solder two resistors, touch controller and buttons by myself on the side when the screen is. Now the pcb that I wrote is working was good. I could flash fw and everything was OK via usb cable. No hot components. Then I've attached shield and put the battery and turn on. It looked like there was no sufficient power so removed battery and when I connected via usb I've noticed that my cp2102 is getting super hot.

zRCrackiiN, 2025-06-05:
> It is recommended to use the original CP2102. In addition, the output voltage of pin 3 of LF33 is 3.3V, and pin 1 input voltage is 5V.
> 
> [image]

what software is this?

silentfox2023, 2025-06-06:
> > It is recommended to use the original CP2102. In addition, the output voltage of pin 3 of LF33 is 3.3V, and pin 1 input voltage is 5V.
> > [image]
> 
> what software is this?

EasyEDA Professional Edition