cifertech/ESP32-DIV #33: Why can't I burn successfully?

Lyf-54, 2025-06-01:


Lyf-54, 2025-06-01:
[image]

LTD284, 2025-06-01:
Use arduino ide to download software to board, if any question ask me

t1nkersec, 2025-06-02:
> Use arduino ide to download software to board, if any question ask me

But if it's an esp32, why couldn't the .bin file just work as a firmware fash?

Lyf-54, 2025-06-02:
> 查看 wiki 以了解正确的燃烧姿势。在项目 https://github.com/cifertech/ESP32-DIV/blob/main/Flash%20File/flash_download_tool_3.9.5.zip 中采用该方法

I'm using flash.,Why is there an error?,Did you burn it successfully?

silentfox2023, 2025-06-02:
Why use "flash_download_tool" at all? There's a better way, using "esptool" is simpler. For example:  
esptool -b 115200 write_flash --flash_size detect 0x8000 ESP32 - DIV.ino.partitions.bin 0x1000 ESP32 - DIV.ino.bootloader.bin 0x10000 ESP32 - DIV.ino.bin

silentfox2023, 2025-06-03:
Actual tests have found that both ESP32U and ESP32S can be successfully programmed. In fact, for ESP32 derivatives including ESP32UE, any module with the code ESP32-D0WD (which can be checked using `esptool flash_id`) is compatible. These modules essentially only differ in chip packaging forms, specifically in whether the module is equipped with an onboard antenna.

In addition, most of the problems with failed programming are due to issues with the PCB. For example, poor soldering, the use of inferior components, etc. In fact, the original design supports automatic programming, which can be confirmed by checking the PCB design drawings. Therefore, please inspect your PCB. Good luck!

Finally, I would like to recommend the Bruce project. This project has a relatively large user base, rapid version updates, and a rich set of features. Its codebase is relatively mature, though its appearance is not as sleek as that of ESP32DIV. It can run on devices such as the M5StickC PLUS2 and T-Embed, eliminating the need for PCB soldering.

[image]
[image]
[image]

t1nkersec, 2025-06-03:
> Actual tests have found that both ESP32U and ESP32S can be successfully burn. In fact, for ESP32 derivatives including ESP32UE, any module with the code ESP32-D0WD (which can be checked using `esptool flash_id`) is compatible. These modules essentially only differ in chip packaging forms, specifically in whether the module is equipped with an onboard antenna.
> 
> In addition, most of the problems with failed burn are due to issues with the PCB. For example, poor soldering, the use of inferior components, etc. In fact, the original design supports automatic programming, which can be confirmed by checking the PCB design drawings. Therefore, please inspect your PCB. Good luck!
> 
> Finally, I would like to recommend the Bruce project. This project has a relatively large user base, rapid version updates, and a rich set of features. Its codebase is relatively mature, It's just not as cool as esp32div. It can run on devices such as the M5StickC PLUS2 and T-Embed, eliminating the need for PCB soldering.
> 
> [image]
> [image]
> [image]

Thank you for your in-depth explanation!!

May I ask if you used the current BOM and Gerber files to build the project? Also did you use a PnP file?

zRCrackiiN, 2025-06-03:
[image]

silentfox2023, 2025-06-04:
> Thank you for your in-depth explanation!!
> 
> May I ask if you used the current BOM and Gerber files to build the project? Also did you use a PnP file?

I used the original author's bill of materials to ensure code compatibility and stability. However, except for the 100μF 50V capacitor, I think a 100μF 16V capacitor should be used instead. After all, I didn't have time to study someone else's code. In the end, I found that there was a problem with the original author's design. I wonder if others have encountered the same issue.

By the way, I do manual soldering of components.

silentfox2023, 2025-06-04:
I used the original author's bill of materials to ensure code compatibility and stability. However, except for the 100μF 50V capacitor, I think a 100μF 16V capacitor should be used instead. After all, I didn't have time to study someone else's code. In the end, I found that there was a problem with the original author's design, it's about the pin conflicts between the CC1101 and NRFL24L01P modules. I wonder if others have encountered the same issue.

My Issue:[https://github.com/cifertech/ESP32-DIV/issues/35](url)

By the way, I do manual soldering of components.

DanielR92, 2025-06-14:
@Lyf-54 something new here? I have the same problem.

wfhstudio, 2025-06-26:
> [image]

Just use 1 bin file (the firmware) and the address is 0

DanielR92, 2025-07-08:
When I flash the bin with adress 0, I get this in the serial output:

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)<\r><\n>
invalid header: 0x75201007<\r><\n>
invalid header: 0x75201007<\r><\n>
invalid header: 0x75201007<\r><\n>
invalid header: 0x75201007<\r><\n>
invalid header: 0x75201007<\r><\n>
invalid header: 0x75201007<\r><\n>
invalid header: 0x75201007<\r><\n>
invalid header: 0x7520ets Jul 29 2019 12:21:46<\r><\n>

BannanaWithZipper, 2025-09-09:
> You must be a beginner. You will understand this when you use the finished product. In fact, the designer Cifer has already started designing a new circuit board. There is a video of his on YouTube titled "I Fixed Everything Wrong with ESP32-DIV", which speaks for itself. Additionally, the CP2102 uses a QFN package, so you must use a hot air gun or a heating platform for soldering. Finally, it should be noted that all of Cifer's designs have a common issue: the pin 1 of the charging management chip TP4056 is left floating. This may cause TP4056 chips produced by some manufacturers to determine that there is a problem with the chip's operating temperature and thus stop working. Forgot to mention that for pins like those of TYPE-C, soldering needs to be done under a microscope. Alternatively, experienced technicians can also solder directly by applying solder with a knife-shaped soldering iron tip.

Yes, but to be able to use the finished product to "understand"... I just need the coordinates to at least the first 3 components, which for me are the hardest to solder by hand. I can manage the rest... already started soldering the others but ruined the boards while trying to solder the usb-c connector.