cifertech/ESP32-DIV #35: **Design Flaws in ESP32DIV - Pin Conflicts Cause Module Initialization Failures & Stability Issues (with Solutions)**

silentfox2023, 2025-06-04:
At present, there are problems in the ESP32DIV design. Since all pins of the CC1101 module and the second NRF24L01P module are shared, including the critical CE (GOD0) and CSN (the GOD2 pin is the CE pin of the first NRF24L01P module), pin conflicts have occurred. This causes the second NRF24L01P to fail to initialize and the CC1101 to operate unstably. According to the circuit principle, it is necessary to redesign the pins of the CC1101 or NRF24L01P to at least ensure that CE and CSN pins are not shared. Alternatively, a switch can be designed to ensure that the CC1101 module and the NRF24L01P module cannot be powered on simultaneously.  

In addition, the same problem exists between the SD module and the third NRF24L01P module. The CS pin of the SD and the CSN pin of the NRF24L01P share GPIO5. Therefore, after inserting the SD card, the NRF24L01P may fail to initialize or operate unstably, although partial fixes may be possible through code.  

 it is recommended that everyone use this cautiously and suggest redesigning the circuit board to ensure usability and stability.

Hey, I also want to hear everyone's suggestions and usage experiences.

[image]

silentfox2023, 2025-06-04:
There are a total of 4 modules using SPI communication, and the SD module also uses SPI communication. How can conflicts be avoided? Currently, I haven't seen anyone do this before. The only idea I have is to distinguish them through different register addresses, but this is only logical. Physically, the Chip Select signals will be transmitted to the two modules simultaneously.

t1nkersec, 2025-06-25:
> There are a total of 4 modules using SPI communication, and the SD module also uses SPI communication. How can conflicts be avoided? Currently, I haven't seen anyone do this before. The only idea I [...]

Do you think it's worth redesigning the PCB and including a switch maybe or does Cifer have a redesign on the way already? It's hard to reach him.

silentfox2023, 2025-06-26:
Designing a separate switch is not as good as redesigning to connect other pins of the ESP32. It can be understood by looking at the circuit diagram. The ESP32 still has at least two idle pins, IO35 and IO39, which can respectively replace the CS pins of the SD module and the second NRF24L01P module. However, the problem is that both boards need to be redesigned and prototyped.