cifertech/ESP32-DIV #36: The nrf24 module is not functioning

Liat583, 2025-06-04:
My self-made ESP32div can start, but the NRF24 module cannot work. The wiring is correct and the module is not damaged

Liat583, 2025-06-05:
> 所有的吗CC1101模块工作正常吗？

CC1101 can be used, but NRF24 cannot be used normally

Liat583, 2025-06-06:
> First, CC1101 and the second NRF24 module use exactly the same pins, so it is normal that the second one cannot be used. Second, the CS pin of the SD module is also shared with the CS pin of the third NRF24 module. Therefore, the original author does not allow modifying the code related to the SD card part, which I think may be the reason. Finally, after flashing my firmware, you can use the serial monitor to check the echo information to see if the initialization is successful.
> 
> [ESP32U-ESP32DIV-PCB-Modify_V1.1.zip](https://github.com/user-attachments/files/20613780/ESP32U-ESP32DIV-PCB-Modify_V1.1.zip)
> 

How should I solve it?

RoToSpLaSh, 2025-06-07:
I am in this situation, I remapped the Touch screen and now it is no longer inverted, I have 2 nrf24l01+ modules connected and working, the third excludes the sd and vice versa, the pcf8574 works perfectly, the pins for the cc1101 are missing, 
    maybe using two esp32s?

RoToSpLaSh, 2025-06-08:
https://github.com/user-attachments/assets/0ae17fed-6714-46b3-90bd-a7aeba492024