cifertech/ESP32-DIV #39: ST7789

PogiPhatzxxx, 2025-06-10:
Make a support for ST7789 display please

zloneotm, 2025-06-10:
I'm working on it right now, so far the driver works with original code. I just followed the wiring diagram accordingly. Currently editing code for smoother touch interface[image]

zloneotm, 2025-06-11:
> > I'm working on it right now, so far the driver works with original code. I just followed the wiring diagram accordingly. Currently editing code for smoother touch interface[image]
> 
> Did you use PCF8574T for the buttons?
> 

I will try to upload the schematic on my git today. 
I used this PCF8574T 

Teyleten Robot PCF8574T PCF8574... https://www.amazon.com/dp/B098B5CGYJ?ref=ppx_pop_mob_ap_share

With SCL to IO22 on ESP32
and SDA to IO21