cifertech/ESP32-DIV #41: Replacing nRF24L01 modules.

Kunelux, 2025-06-10:
This is not really an issue, but a doubt. I was wondering wheter it would be possible to replace the nRF24L01 PA+LNA modules with some other more powerful ones like the E01-ML01DP5 or E01-2G4M27D. Thanks a lot.

Kunelux, 2025-06-10:
> [image]
> 
> 2 e01-2g4m27d and 1 nrf24l01

That's helpful! btw, I use 10uF capacitors on the nRF24L01 but would it be better to use 100uF capacitors with the E01-2G4M27D, or just stick to 10uF?

RoToSpLaSh, 2025-06-10:
> > [image]
> > 
> > 2 e01-2g4m27d and 1 nrf24l01
> 
> That's helpful! btw, I use 10uF capacitors on the nRF24L01 but would it be better to use 100uF capacitors with the E01-2G4M27D, or just stick to 10uF?

I use the same ones for both form types

Kunelux, 2025-06-10:
> > > [image]
> > > 2 e01-2g4m27d and 1 nrf24l01
> > 
> > 
> > That's helpful! btw, I use 10uF capacitors on the nRF24L01 but would it be better to use 100uF capacitors with the E01-2G4M27D, or just stick to 10uF?
> 
> I use the same ones for both form types

Oh alright, I read somewhere that because of those being more "powerful" it would be better to use a 100uF but its whatever. Would it be an issue for you to send a pic of your project?

Kunelux, 2025-06-10:
> I followed the connection scheme that you find on the author's page plus some changes to the code that you can find in the various comments that I left.

Understood, thanks a lot. Just so you know, I was thinking about using an amplifier like these and power it through another battery, maybe you could add it to your project. Thanks for the help![

[image]

RoToSpLaSh, 2025-06-10:
This amplifier is for subghz rf frequencies to be used with the cc1101 module, I have it too but I use it for other projects

Kunelux, 2025-06-10:
> This amplifier is for subghz rf frequencies to be used with the cc1101 module, I have it too but I use it for other projects

Yeah, I didn't mean this one but something similar. Do you think it's worth it for the CC1101?

RoToSpLaSh, 2025-06-10:
> > This amplifier is for subghz rf frequencies to be used with the cc1101 module, I have it too but I use it for other projects
> 
> Yeah, I didn't mean this one but something similar. Do you think it's worth it for the CC1101? 

there are better ones

[image]

silentfox2023, 2025-06-11:
> This is not really an issue, but a doubt. I was wondering wheter it would be possible to replace the nRF24L01 PA+LNA modules with some other more powerful ones like the E01-ML01DP5 or E01-2G4M27D. Thanks a lot.

The original power supply design can't keep up, which many people have thought of. Unfortunately, the transmitting current of E01-2G4M27D is 720mA, and this is just for one module. When three NRF24L01P modules are used simultaneously, the ESP32DIV will directly go blackscreen or restart. In addition, E01-ML01DP5 can be tried, with a transmitting current of 130mA, but the situation is estimated to be not optimistic. It is also feasible. Try replacing LF33.

Kunelux, 2025-06-11:
> > This is not really an issue, but a doubt. I was wondering wheter it would be possible to replace the nRF24L01 PA+LNA modules with some other more powerful ones like the E01-ML01DP5 or E01-2G4M27D. Thanks a lot.
> 
> The original power supply design can't keep up, which many people have thought of. Unfortunately, the transmitting current of E01-2G4M27D is 720mA, and this is just for one module. When three NRF24L01P modules are used simultaneously, the ESP32DIV will directly go blackscreen or restart. In addition, E01-ML01DP5 can be tried, with a transmitting current of 130mA, but the situation is estimated to be not optimistic. Alternatively, you can try replacing the LF33.
> 

Maybe using an additional battery could work, even if a redesign of the pcb would have to be done, or completely remade. Also, the space needed would be a problem.