cifertech/ESP32-DIV #43: 3D Print?

zloneotm, 2025-06-11:
Anyone develop a case/holder?

zloneotm, 2025-06-14:
> [image]

That was the one I saw in the one of the issues pull, amazing work! 
Stl in your git by chance?

t1nkersec, 2025-07-02:
I think if someone modifies this case it could work. The PCB takes a lot of inspiration from it! 

https://www.thingiverse.com/thing:4260973

zloneotm, 2025-07-02:
[image]

t1nkersec, 2025-07-02:
> > I think if someone modifies this case it could work. The PCB takes a lot of inspiration from it!
> > https://www.thingiverse.com/thing:4260973
> 
> I actually developed one for my diy pcb [image]

Good stuff!! I like the H2 design but it needs quite a few modifications for the extra SMA antennas!