cifertech/ESP32-DIV #48: Button 0: Pressed

X8OCT, 2025-06-25:
Please tell me what the problem might be. The buttons don't work, the display is OK, but the DIV itself works very slowly. The following can be seen in the log

> 14:54:28.337 -> ets Jul 29 2019 12:21:46
> 14:54:28.337 -> 
> 14:54:28.337 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
> 14:54:28.337 -> configsip: 0, SPIWP:0xee
> 14:54:28.337 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
> 14:54:28.337 -> mode:DIO, clock div:1
> 14:54:28.337 -> load:0x3fff0030,len:1344
> 14:54:28.337 -> load:0x40078000,len:13964
> 14:54:28.337 -> load:0x40080400,len:3600
> 14:54:28.337 -> entry 0x400805f0
> 14:54:28.919 -> E (571) ADC: adc1_lock_release(419): adc1 lock release called before acquire
> 14:54:28.919 -> E (572) ADC: adc1_lock_release(419): adc1 lock release called before acquire
> 14:54:28.919 -> E (574) ADC: adc1_lock_release(419): adc1 lock release called before acquire
> 14:54:28.919 -> E (581) ADC: adc1_lock_release(419): adc1 lock release called before acquire
> 14:54:28.919 -> E (588) ADC: adc1_lock_release(419): adc1 lock release called before acquire
> 14:54:28.919 -> E (594) ADC: adc1_lock_release(419): adc1 lock release called before acquire
> 14:54:28.919 -> E (601) ADC: adc1_lock_release(419): adc1 lock release called before acquire
> 14:54:28.952 -> E (608) ADC: adc1_lock_release(419): adc1 lock release called before acquire
> 14:54:28.952 -> E (615) ADC: adc1_lock_release(419): adc1 lock release call�x@before acquire
> 14:54:28.952 -> E (621) ADC: adc1_lock_release(419): adc1 lock release called before acquire
> 14:54:31.549 -> ==================================
> 14:54:31.549 -> ESP32-DIV                         
> 14:54:31.549 -> Developed by: CiferTech           
> 14:54:31.549 -> Version:      1.1.0               
> 14:54:31.549 -> Contact:      cifertech@gmail.com 
> 14:54:31.549 -> GitHub:       github.com/cifertech
> 14:54:31.549 -> Website:      CiferTech.net       
> 14:54:31.549 -> ==================================
> 14:54:33.546 -> Button 0: Pressed
> 14:54:33.546 -> Button 1: Pressed
> 14:54:33.546 -> Button 2: Pressed
> 14:54:33.546 -> Button 3: Pressed
> 14:54:33.546 -> Button 4: Pressed
> 14:54:33.546 -> Button 5: Pressed
> 14:54:33.546 -> Button 6: Pressed
> 14:54:33.546 -> Button 7: Pressed

X8OCT, 2025-06-27:
Its nrf24 problem :(

X8OCT, 2025-06-27:
[image]

closed 2025-06-28 by X8OCT

naf9000, 2025-07-01:
alr, ill try that, ill be back in just a minute

X8OCT, 2025-07-01:
it may differ from mine, better look in the datasheet of ur PCF8574

naf9000, 2025-07-01:
yeah im pretty sure 0x20 is correct
what should i do then?

naf9000, 2025-07-01:
afaik all needed libraries are installed and everything is soldered correctly

X8OCT, 2025-07-01:
PCF8574T and xpt2046 soldered with a key?

X8OCT, 2025-07-01:
Have you tried running it without a shield? Will it be possible to show the connected libraries? And a photo of the board?

naf9000, 2025-07-01:
[image]
Flashing settings^

naf9000, 2025-07-01:
[image]

X8OCT, 2025-07-01:
[image]

X8OCT, 2025-07-01:
8774 6pin looks like not soldering 
and I don’t quite understand, is it 10 Ohm or jumpers?

X8OCT, 2025-07-01:
[image]

naf9000, 2025-07-01:
if i replaced the jumpers with 10R resistors would it theoretically work?

X8OCT, 2025-07-01:
jumpers are not scary since this is the display backlight line, but it would be nice to put

naf9000, 2025-07-01:
still encountering the same issues
also replaced the bridges with 10ohm resistors

X8OCT, 2025-07-01:
I'd also like to take a look at the display cable and go through all the 8574 pins. It would also be good to solder the 8050 near the jumpers

naf9000, 2025-07-01:
even still, i checked the connections with my multimeter and everything seems to be ok

X8OCT, 2025-07-01:
try like this with solder paste

[image]

naf9000, 2025-07-01:
i moved the display back behind the line, still experiencing same issues

naf9000, 2025-07-01:
can you explain what to do?

naf9000, 2025-07-01:
i originally soldered everything with solder paste
i may purchase another display to test
capacitors do fall into holes as they should

X8OCT, 2025-07-02:
soldering paste is not quite what you need, it is better to solder contacts using flux or rosin
I admit that there is not good contact everywhere, you can carefully move the legs with tweezers on the same 8574

naf9000, 2025-07-02:
alright, i'll try that

naf9000, 2025-07-16:
Hello,
I have replaced the esp32, PCF8574T, and XPT2046 components.
Unfortunately, while removing the display, the ribbon cable tore. I plan to get an external display soon for testing.
I am also still experiencing the same error.